edgeimpulse/firmware-pi-rp2xxx #1: Serial comms hang in Win10

ShawnHymel, 2022-03-29:
I'm trying to connect the RPi Pico to Windows 10. I followed the guide to upload the latest firmware .uf2 file to the Pico. When I run `edge-impulse-daemon`, it hangs forever trying to connect to the COM port. 

[image]

I also cannot connect via PuTTY--the same thing happens: it hangs forever.

Note that the "hello usb" firmware does work. If I upload the .uf2 file found here (https://forums.raspberrypi.com/viewtopic.php?t=314921#p1883936) to the Pico, I can see "Hello, World" printed to the console (e.g. connected via Putty at 115200 baud rate).

AIWintermuteAI, 2022-04-19:
Hi, Shawn! 
We'll investigate the issue. By the way, does it work on Mac for you? It could an issue with Windows drivers...

ShawnHymel, 2022-04-19:
@AIWintermuteAI Just tried it--it hangs on my Mac, too.

[image]

KevWal, 2022-04-19:
Hi, I have a similar issue that I posted on the forum.  Copied here incase the additional details are useful.

I have a brand new Pi Pico.

Flashed with rp2-pico-20220117-v1.18.uf2 from: [MicroPython - Python for microcontrollers](https://micropython.org/download/rp2-pico/)
I get a Com5 USB\VID_2E8A&PID_0005&MI_00\7&381503f8&0&0000 in Zadig showing as Board CDC (Interface 0) using usbser (v10.0.19041.1202) and putty gives me a connection and a python prompt - so the board and the usb / serial connection can work.

Flashing with ei_rp2040_firmware.uf2 from https://docs.edgeimpulse.com/docs/development-boards/raspberry-pi-rp2040
I get a Com4 USB\VID_2E8A&PID_000A&MI_00\7&207451d&0&0000 in Zadig showing as Board CDC (Interface 0) using usber (V1.0.0.0) and putty does not connect on Com4 – it just hangs and I have to unplug the pico to get Putty to release the port.

Trying to update the driver using Zadig 2.7 says it was successful but has no effect, even after a reboot still shows as v1.0.0.0 and putty still hangs,

Replacing the driver with winusb, and then back to USB Serial (CDC) it still goes back to v1.0.0.0

Finally in Device Manager, using “Update Driver” and Browse I managed to get the version up to v10.0.19041.1202 – but still no putty connection

I also tried a Chrome “Connect using WebUSB”, which shows a ‘Board CDC (Com 4) - paired’, but I get a “Failed to connect to device” error straight away or sometime Chrome just hanging and not doing anything.

I have also tried on a different windows machine, and I still get no putty connection.

Thanks very much
Kevin

AIWintermuteAI, 2022-04-20:
Okay, so I can partially reproduce the problem:
1) If I connect Pico to PC and immediately execute ```edge-impulse-daemon --verbose```, the device is connected normally.
[image]
2) However, for consecutive executions of  ```edge-impulse-daemon --verbose```, there is a chance of not being able to read config from device - the issue you described. Plugging out and plugging in Pico fixes that.
I tested connection on Ubuntu 20.04 and MacOS. 
Can you confirm that the problem is intermittent?

ShawnHymel, 2022-04-20:
@AIWintermuteAI yes, I confirmed that it is intermittent on my Mac. I plugged it in and could connect a few times. On about the third or fourth time, it loses serial connection to the Pico.

[image]

KevWal, 2022-04-20:
For me, on Windows, it is permanent,  I have never been able to connect with ei_rp2040_firmware.uf2 uploaded and booted.

AIWintermuteAI, 2022-04-28:
Okay, can you test this firmware - it works for me on Windows 10. Before the change (latest version of tinyUSB in pico-sdk) I could reproduce your problem. Now I can connect.
[image]

KevWal, 2022-04-28:
Thank you, will do next week, am away for a long weekend.  Cheers, Kev

KevWal, 2022-05-01:
> Okay, can you test this firmware - it works for me on Windows 10. Before the change (latest version of tinyUSB in pico-sdk) I could reproduce your problem. 

And first PC I tried it one (which didn't work before), it now works :)

[image]

Thanks very much
Kevin

RickorDD, 2022-06-14:
I had the same problem with Arduino RP2040. I have install Arduino IDE 2.0 RC7 with the deprecated Drivers (1.3.2) RP2040 Support and the new RP2040 Drivers. I removed the deprecated Hardware Package in Folder C:\Users\***\AppData\Local\Arduino15\packages\arduino\hardware then the RP2040 was connected.

ShawnHymel, 2022-07-12:
From a user, not sure if it helps ([forum post](https://forum.edgeimpulse.com/t/errors-with-officially-supported-boards-esp32-cam-and-rp2040-pi-pico/4829)):

> Hi! I found the solution to this annoying problem.
> I needed to compile the firmware locally and in the PICO-SDK >src>rp2_common>pico_stdio_usb>include>pico>stdio_usb.h I just needed to set the variable:
> 
> PICO_STDIO_USB_ENABLE_RESET_VIA_VENDOR_INTERFACE = 0
> (default is set to 1)
> This error is happening on Win 10 and 11

closed 2022-07-25 by AIWintermuteAI

AIWintermuteAI, 2022-07-25:
I just merged latest upgrade for RP2040 to master. Can someone try after it is deployed to production (tomorrow at latest)?

ShawnHymel, 2022-08-02:
@AIWintermuteAI I downloaded the firmware from https://docs.edgeimpulse.com/docs/development-platforms/officially-supported-mcu-targets/raspberry-pi-rp2040 and it still hangs. Is that firmware up-to-date?

[image]

ShawnHymel, 2022-08-02:
Nevermind...I tried the latest .uf2 from https://github.com/edgeimpulse/firmware-pi-rp2040/tree/main/build and it seems to be working now. I'm guessing that the one on the docs page has not been updated.

AIWintermuteAI, 2022-08-03:
It should be latest firmware there by now. Can you try again? The size of the .uf2 file should be 1.1 Mb.

ShawnHymel, 2022-08-03:
Confirmed--download is 1.1 MB and it works!