edgeimpulse/firmware-pi-rp2xxx #2: Enable running impulse continuously in the AT handler

ShawnHymel, 2022-07-19:
The AT command to run an impulse continuously is [disabled](https://github.com/edgeimpulse/firmware-pi-rp2040/blob/7771eef0de4e13d8037513e9ec224b171d838033/edge-impulse/ingestion-sdk-platform/raspberry-rp2040/ei_at_handlers.cpp#L581):

```
//at->register_command(RUNIMPULSECONT, "Run the impulse continuously", nullptr, nullptr, at_run_nn_continuous_normal, AT_RUNIMPULSECONT_ARGS);
```

A user is requesting to have it enabled: https://forum.edgeimpulse.com/t/continuous-mode-on-rp2040-disabled/4988

AIWintermuteAI, 2022-07-20:
Hi, @ShawnHymel !
Thanks for the information! We’re in the process of review of PR that would allow using microphone on Arduino RP2040 Connect and also will add continuous inference support - only for microphone and camera though, not generic fusion sensors. The user asked about microphone, so it will satisfy his use case.
Stay tuned.

AIWintermuteAI, 2022-07-29:
v1.4.17 of Edge Impulse includes latest update - microphone on Arduino RP2040 Connect and continious inference for it should be available now.
https://github.com/edgeimpulse/firmware-pi-rp2040/blob/8acd9fa511977c5fd9f44cc2bd1c68e549220844/edge-impulse/inference/ei_run_audio_impulse.cpp#L127

closed 2022-07-29 by AIWintermuteAI