edgeimpulse/firmware-pi-rp2xxx #4: Unable to build firmware-pi-rp2040 on my mac laptop

jakeoldboy, 2022-09-07:
All:

I'm unable to build Edge Impulse's firmware-pi-rp2040 (for my new Arduino Nano rp2040 Connect) on my mac laptop after doing the following:

cd ~/workspace
mkdir arduino
cd arduino

git clone https://github.com/edgeimpulse/firmware-pi-rp2040.git
git clone https://github.com/raspberrypi/pico-sdk.git

export PATH=~/workspace/arduino/pico-sdk:${PATH}:
brew install cmake
brew install gcc-arm-none-eabi
brew tap ArmMbed/homebrew-formulae
brew install arm-none-eabi-gcc

which cmake
which arm-none-eabi-gcc

cd firmware-pi-rp2040

mkdir build && cd build

export PICO_SDK_PATH=~workspace/arduino/pico-sdk
cmake ..
make -j4

16:41 $ cmake ..
Using PICO_SDK_PATH from environment ('~/workspace/arduino/pico-sdk')
PICO_SDK_PATH is ~/workspace/arduino/pico-sdk
Defaulting PICO_PLATFORM to rp2040 since not specified.
Defaulting PICO platform compiler to pico_arm_gcc since not specified.
-- Defaulting build type to 'Release' since not specified.
PICO compiler is pico_arm_gcc
Setting FREERTOS_KERNEL_PATH to '~/workspace/arduino/firmware-pi-rp2040/ThirdParty/FreeRTOS-Kernel' found relative to enclosing project
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/local/bin/arm-none-eabi-gcc
Build type is Release
Defaulting PICO target board to pico since not specified.
Using board configuration from ~/workspace/arduino/pico-sdk/src/boards/include/boards/pico.h
-- Found Python3: /usr/local/Frameworks/Python.framework/Versions/3.9/bin/python3.9 (found version "3.9.2") found components: Interpreter
CMake Warning at ~/workspace/arduino/pico-sdk/src/rp2_common/tinyusb/CMakeLists.txt:10 (message):
  TinyUSB submodule has not been initialized; USB support will be unavailable

  hint: try 'git submodule update --init' from your SDK directory
  (~/workspace/arduino/pico-sdk).

-- Configuring done
-- Generating done
-- Build files have been written to: ~/workspace/arduino/firmware-pi-rp2040/build

base) ✔   ~/workspace/arduino/firmware-pi-rp2040/build [main|✔]
16:42 $ ls -lt
total 3408
drwxr-xr-x  17 jpingel  staff      544 Sep  7 16:42 CMakeFiles
drwxr-xr-x   8 jpingel  staff      256 Sep  7 16:42 pico-sdk
drwxr-xr-x   5 jpingel  staff      160 Sep  7 16:42 FREERTOS_KERNEL
-rw-r--r--   1 jpingel  staff     1869 Sep  7 16:42 cmake_install.cmake
-rw-r--r--   1 jpingel  staff  1713746 Sep  7 16:42 Makefile
-rw-r--r--   1 jpingel  staff    21035 Sep  7 16:42 CMakeCache.txt
drwxr-xr-x   4 jpingel  staff      128 Sep  7 16:42 elf2uf2
drwxr-xr-x   4 jpingel  staff      128 Sep  7 16:42 pioasm
drwxr-xr-x   3 jpingel  staff       96 Sep  7 16:42 generated
(base) ✔   ~/workspace/arduino/firmware-pi-rp2040/build [main|✔]

16:42 $ make -j4
Scanning dependencies of target bs2_default
[  0%] Creating directories for 'PioasmBuild'
[  0%] Creating directories for 'ELF2UF2Build'
[  0%] Building ASM object pico-sdk/src/rp2_common/boot_stage2/CMakeFiles/bs2_default.dir/compile_time_choice.S.obj
[  0%] No download step for 'ELF2UF2Build'
[  0%] No download step for 'PioasmBuild'
[  0%] No update step for 'PioasmBuild'
[  0%] No update step for 'ELF2UF2Build'
[  1%] Linking ASM executable bs2_default.elf
[  1%] No patch step for 'PioasmBuild'
[  1%] No patch step for 'ELF2UF2Build'
[  1%] Performing configure step for 'PioasmBuild'
[  1%] Performing configure step for 'ELF2UF2Build'
loading initial cache file ~/workspace/arduino/firmware-pi-rp2040/build/pioasm/tmp/PioasmBuild-cache-Release.cmake

[  1%] Built target bs2_default
[  1%] Generating bs2_default.bin
[  1%] Generating bs2_default_padded_checksummed.S
[  1%] Built target bs2_default_padded_checksummed_asm
-- The C compiler identification is AppleClang 13.1.6.13160021
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting CXX compiler ABI info
-- The CXX compiler identification is AppleClang 13.1.6.13160021
-- Detecting C compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: ~/workspace/arduino/firmware-pi-rp2040/build/pioasm
[  1%] Performing build step for 'PioasmBuild'
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info

[ 10%] Building CXX object CMakeFiles/pioasm.dir/main.cpp.o
[ 20%] Building CXX object CMakeFiles/pioasm.dir/pio_assembler.cpp.o
[ 30%] Building CXX object CMakeFiles/pioasm.dir/pio_disassembler.cpp.o

-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done

-- Build files have been written to: ~/workspace/arduino/firmware-pi-rp2040/build/elf2uf2

[  1%] Performing build step for 'ELF2UF2Build'
[ 50%] Building CXX object CMakeFiles/elf2uf2.dir/main.cpp.o
[ 40%] Building CXX object CMakeFiles/pioasm.dir/gen/lexer.cpp.o
[100%] Linking CXX executable elf2uf2
[100%] Built target elf2uf2
[  1%] No install step for 'ELF2UF2Build'
[  1%] Completed 'ELF2UF2Build'
[  1%] Built target ELF2UF2Build
[ 50%] Building CXX object CMakeFiles/pioasm.dir/gen/parser.cpp.o
[ 60%] Building CXX object CMakeFiles/pioasm.dir/c_sdk_output.cpp.o
[ 70%] Building CXX object CMakeFiles/pioasm.dir/python_output.cpp.o
[ 80%] Building CXX object CMakeFiles/pioasm.dir/hex_output.cpp.o
[ 90%] Building CXX object CMakeFiles/pioasm.dir/ada_output.cpp.o
[100%] Linking CXX executable pioasm
[100%] Built target pioasm
[  1%] No install step for 'PioasmBuild'
[  2%] Completed 'PioasmBuild'
[  2%] Built target PioasmBuild
[  3%] Generating pdm.pio.h
[  3%] Built target ei_rp2040_firmware_pdm_pio_h
Scanning dependencies of target ei_rp2040_firmware
[  3%] Building CXX object CMakeFiles/ei_rp2040_firmware.dir/edge-impulse-sdk/dsp/image/processing.cpp.obj
[  3%] Building CXX object CMakeFiles/ei_rp2040_firmware.dir/src/main.cpp.obj
[  3%] Building CXX object CMakeFiles/ei_rp2040_firmware.dir/edge-impulse-sdk/classifier/ei_run_classifier_c.cpp.obj
[  3%] Building CXX object CMakeFiles/ei_rp2040_firmware.dir/edge-impulse-sdk/dsp/dct/fast-dct-fft.cpp.obj

~/workspace/arduino/firmware-pi-rp2040/src/main.cpp:24:10: fatal error: tusb.h: No such file or directory
   24 | #include "tusb.h"
      |          ^~~~~~~~
compilation terminated.

make[2]: *** [CMakeFiles/ei_rp2040_firmware.dir/src/main.cpp.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/ei_rp2040_firmware.dir/all] Error 2
make: *** [all] Error 2
(base) ✘-2   ~/workspace/arduino/firmware-pi-rp2040/build [main|✔]
16:43 $

 ....

So, where does the tusb.h file supposed to come from?

Thanks,
John Pingel

AIWintermuteAI, 2022-09-08:
Interesting. One possibility is that PICO SDK version is wrong.
Can you try 
```
git clone --recurse-submodules --depth 1 https://github.com/raspberrypi/pico-sdk
cd pico-sdk/lib/tinyusb && git checkout 6ec5174d8b24d1f01a443c74a0a3dbee00523efc
```
to get specific commit of pico sdk we use internally. Replace your current pico sdk with it. 
+ remember to clean the build folder and rerun cmake.

jakeoldboy, 2022-09-08:
Thanks for your quick reply.  I’ll try your suggestion below.  I’ll also journal the commands that I use when I try rebuilding the firmware in case I still can’t get it to work.

Thanks,
John Pingel

jakeoldboy, 2022-09-15:
The firmware built successfully with you suggested corrections to the build commands, that is:
    git clone --recurse-submodules --depth 1 https://github.com/raspberrypi/pico-sdk
    cd pico-sdk/lib/tinyusb && git checkout 6ec5174d8b24d1f01a443c74a0a3dbee00523efc
 was also able to copy the newly built firmware to my Arduino Nano RP2040 Connect device.

After installing the edgeimpulse CLI and starting the edge-impulse-daemon, I was able to login to edgeimpulse.com in the Chrome browser and with the edge-impulse-daemon.  I then was able to collect data samples from my device that is connected to my Mac laptop.  The sensor data collected while doing the continuous motion gestures looks reasonable for the type of gestures performed; however, the values in the sample datasets has the accelerometer swapped with that of the gyro.

The constants in the source file:
    ./firmware-pi-rp2040/ThirdParty/Wire/src/Wire.h
are:

// Wire
#define SDA  12
#define SCL  13

#define PIN_WIRE1_SDA  6
#define PIN_WIRE1_SCL  7

What if any other constants may need to be corrected for the accelerometer and gyro values being correct?

Thanks,
John P.

jakeoldboy, 2022-09-15:
[image]

AIWintermuteAI, 2022-09-15:
Since your original issue was resolved, could you please create another issue for your other problem? Each issue is not a place to help one user, but a place to resolve a specific problem. Once the problem is resolved, the issue should be closed.
https://ben.balter.com/2014/11/06/rules-of-communicating-at-github/#9-keep-discussions-logically-distinct

closed 2022-09-15 by AIWintermuteAI