edgeimpulse/firmware-pi-rp2xxx #5: Arduino Nano rp2940 accel and gyro values appear to be swapped

jakeoldboy, 2022-09-15:
After installing the edgeimpulse CLI and starting the edge-impulse-daemon, I was able to login to edgeimpulse.com in the Chrome browser and with the edge-impulse-daemon. I then was able to collect data samples from my device that is connected to my Mac laptop. The sensor data collected while doing the continuous motion gestures looks reasonable for the type of gestures performed; however, the values in the sample datasets has the accelerometer swapped with that of the gyro.

The constants in the source file:
./firmware-pi-rp2040/ThirdParty/Wire/src/Wire.h
are:

// Wire
#define SDA 12
#define SCL 13

#define PIN_WIRE1_SDA 6
#define PIN_WIRE1_SCL 7

What if any other constants may need to be corrected for the accelerometer and gyro values being correct?

Thanks,
John P.

AIWintermuteAI, 2022-09-15:
Hello!
In your screenshot I do not see values in the sample datasets has the accelerometer swapped with that of the gyro. What made you to come to the conclusion they are swapped?

jpingel-clgx, 2022-09-15:
The large amplitude of the yellow and orange signals corresponds to me moving the Adruino Nano RP2040 board from side to side on my laptop. The legend of the graph shows the orange and yellow lines belonging to the gyrX, gyrY, gyrZ sensors.  The red, green, blue lines for the accX, accY, accZ sensors does have about the same periodicity but a much, much smaller amplitude.   Maybe the gyro sensor values are much greater that that of the accelerometer (i.e. I could be wrong).

AIWintermuteAI, 2022-09-30:
Yes, accX, accY, accZ are in m/s2 units, while gyroscope is in deg/s. That's why the numbers for gyroscope are much larger than for accelerometer. They do not appear swapped.
I think your original question was answered. I'll keep the issue open for a while, since I want to discuss scaling of gyroscope measurements internally - it might make more sense to scale it or use radians instead of degrees per second.

closed 2024-07-08 by AIWintermuteAI