edgeimpulse/firmware-pi-rp2xxx #13: Error compiling firmware sdk 2.1.0 and ubuntu 24.04

carlosdelfino, 2024-12-29:
I am trying to compile the Firmware and I get the error message below, I am using Ubuntu 24.04 and SDK 2.1.0, what can I do to fix such issue?

```
/home/carlosdelfino/.pico-sdk/toolchain/13_3_Rel1/bin/arm-none-eabi-g++ -DARM_MATH_LOOPUNROLL -DCFG_TUSB_DEBUG=0 -DCFG_TUSB_MCU=OPT_MCU_RP2040 -DCFG_TUSB_OS=OPT_OS_PICO -DEIDSP_QUANTIZE_FILTERBANK=0 -DFREERTOS_ENABLED=1 -DFREE_RTOS_KERNEL_SMP=1 -DLIB_BOOT_STAGE2_HEADERS=1 -DLIB_FREERTOS_KERNEL=1 -DLIB_PICO_ATOMIC=1 -DLIB_PICO_BIT_OPS=1 -DLIB_PICO_BIT_OPS_PICO=1 -DLIB_PICO_CLIB_INTERFACE=1 -DLIB_PICO_CRT0=1 -DLIB_PICO_CXX_OPTIONS=1 -DLIB_PICO_DIVIDER=1 -DLIB_PICO_DIVIDER_HARDWARE=1 -DLIB_PICO_DOUBLE=1 -DLIB_PICO_DOUBLE_PICO=1 -DLIB_PICO_FIX_RP2040_USB_DEVICE_ENUMERATION=1 -DLIB_PICO_FLASH=1 -DLIB_PICO_FLOAT=1 -DLIB_PICO_FLOAT_PICO=1 -DLIB_PICO_INT64_OPS=1 -DLIB_PICO_INT64_OPS_PICO=1 -DLIB_PICO_MALLOC=1 -DLIB_PICO_MEM_OPS=1 -DLIB_PICO_MEM_OPS_PICO=1 -DLIB_PICO_MULTICORE=1 -DLIB_PICO_NEWLIB_INTERFACE=1 -DLIB_PICO_PLATFORM=1 -DLIB_PICO_PLATFORM_COMPILER=1 -DLIB_PICO_PLATFORM_PANIC=1 -DLIB_PICO_PLATFORM_SECTIONS=1 -DLIB_PICO_PRINTF=1 -DLIB_PICO_PRINTF_PICO=1 -DLIB_PICO_RUNTIME=1 -DLIB_PICO_RUNTIME_INIT=1 -DLIB_PICO_STANDARD_BINARY_INFO=1 -DLIB_PICO_STANDARD_LINK=1 -DLIB_PICO_STDIO=1 -DLIB_PICO_STDIO_UART=1 -DLIB_PICO_STDIO_USB=1 -DLIB_PICO_STDLIB=1 -DLIB_PICO_SYNC=1 -DLIB_PICO_SYNC_CRITICAL_SECTION=1 -DLIB_PICO_SYNC_MUTEX=1 -DLIB_PICO_SYNC_SEM=1 -DLIB_PICO_TIME=1 -DLIB_PICO_TIME_ADAPTER=1 -DLIB_PICO_UNIQUE_ID=1 -DLIB_PICO_UTIL=1 -DPICO_32BIT=1 -DPICO_BOARD=\"pico_w\" -DPICO_BUILD=1 -DPICO_CMAKE_BUILD_TYPE=\"Release\" -DPICO_COPY_TO_RAM=0 -DPICO_CXX_ENABLE_EXCEPTIONS=0 -DPICO_NO_FLASH=0 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -DPICO_RP2040=1 -DPICO_RP2040_USB_DEVICE_UFRAME_FIX=1 -DPICO_TARGET_NAME=\"ei_rp2040_firmware\" -DPICO_USE_BLOCKED_RAM=0 -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/build -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/. -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/./classifer -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/./tflite-model -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/./model-parameters -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/third_party/ruy -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/third_party/gemmlowp -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/third_party/flatbuffers/include -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/third_party -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/tensorflow -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/dsp -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/classifier -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/anomaly -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/CMSIS/NN/Include -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/CMSIS/DSP/PrivateInclude -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/CMSIS/DSP/Include -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/CMSIS/Core/Include -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse-sdk/porting -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse/ingestion-sdk-c -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse/ingestion-sdk-platform/raspberry-rp2040 -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse/ingestion-sdk-platform/sensors -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse/mbedtls_hmac_sha256_sw -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse/inference -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/edge-impulse/QCBOR/inc -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/firmware-sdk -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/firmware-sdk/at-server -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/Seeed_Arduino_UltrasonicRanger -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/Wire/src -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/Arduino_LSM6DSOX/src -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/rp2040_DHT11_lib -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/PDM/src/include -I/home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_atomic/include -I/home/carlosdelfino/.pico-sdk/sdk/2.1.0/lib/tinyusb/src -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/FreeRTOS-Kernel/portable/ThirdParty/GCC/RP2040/include -I/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/ThirdParty/FreeRTOS-Kernel/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/pico_stdlib_headers/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_gpio/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/pico_base_headers/include -isystem /home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/build/generated/pico_base -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/boards/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2040/pico_platform/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2040/hardware_regs/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_base/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_platform_compiler/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_platform_panic/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_platform_sections/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2040/hardware_structs/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/hardware_claim/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_sync/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_sync_spin_lock/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_irq/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_uart/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_resets/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_clocks/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_pll/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_vreg/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_watchdog/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_ticks/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_bootrom/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/boot_picoboot_headers/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/boot_bootrom_headers/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_boot_lock/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_flash/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/pico_time/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/hardware_timer/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/pico_sync/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/common/pico_util/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_common/pico_time_adapter/include -isystem /home/carlosdelfino/.pico-sdk/sdk/2.1.0/src/rp2_c [... cut ...]
      |                      ^~~~~~~~~~~~~~~~~~~~
      |                      PICO_DEFAULT_SPI
[35/906] Building CXX object CMakeFiles/ei_rp2040_firmware.dir/edge-impulse-sdk/porting/raspberry/ei_classifier_porting.cpp.o
ninja: build stopped: subcommand failed.

 *  The terminal process "/home/carlosdelfino/.pico-sdk/ninja/v1.12.1/ninja '-C', '/home/carlosdelfino/workspace/EmbarcaTech/EdgeImpulse/firmware-pi-rp2040/build'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

AIWintermuteAI, 2024-12-30:
Hello!

Internally we use 1.5.1 for building Pico firmware
https://github.com/raspberrypi/pico-sdk/releases/tag/1.5.1

carlosdelfino, 2024-12-30:
Dear friend

I tested with all versions of the SDK, including 1.5.1 which gives the same error, I was careful to clean cmake and recreate the make files.

I am using VSCode with the official Raspberry Pico extension.

carlosdelfino, 2024-12-31:
For this LED error, I corrected the code and added the directives so that it uses the correct LED in the case of pico-w.

As soon as I find the other corrections, I will gather everything and send it to you for analysis in a pull request.

I have already spent many hours trying to solve this problem and nothing has worked.

AIWintermuteAI, 2025-01-02:
I was able to compile firmware locally just now. One major difference I see is the compiler version, this is likely the source of time_t issue:
```
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
```
The easiest solution for you might be to quickly spin a Docker container, install the 10.3.1 compiler there and try with it? 

Also, what is the reason you want to compile the firmware? The default workflow for most of the users is to use a pre-compiled firmware, did that work for you?

carlosdelfino, 2025-01-02:
I am using a custom board provided by the Technical School, defined by the Brazilian government's educational project. I cannot physically change the ports, I need to change them internally in the software.

In addition, I need to use as many resources as possible that the educational kit offers, such as WiFi.

This is my first contact with Pico W, and the course deadline is short.

To compile the code, I am using VSCode, I will try without it.

carlosdelfino, 2025-01-02:
I followed step by step what is in [README.md](https://github.com/edgeimpulse/firmware-pi-rp2040/blob/main/README.md) and the same error occurred.

AIWintermuteAI, 2025-01-02:
What is the output of `arm-none-eabi-gcc --version`?

carlosdelfino, 2025-01-02:
carlosdelfino@estudio-6b:~/workspace/EmbarcaTech/firmware-pi-rp2040$  arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:13.2.rel1-2) 13.2.1 20231009
Copyright (C) 2023 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

carlosdelfino, 2025-01-02:
vscode doesn't use the same version, and in ubuntu 24.04 it uses another version, and as you said you need to use version 10.3.1 which is relatively old.

I'll find a way to install 10.3.1, and configure it especially for this case.

AIWintermuteAI, 2025-01-02:
As I mentioned in https://github.com/edgeimpulse/firmware-pi-rp2040/issues/13#issuecomment-2567858393
you need arm-none-eabi-gcc 10.3.x.

The best way for you to use 10.3.x and avoid messing with system compiler is to use docker.

carlosdelfino, 2025-01-02:
Could you tell me where I can get a docker image with this version?

I did a quick search on Google, and I haven't found it yet.

AIWintermuteAI, 2025-01-02:
Just get ubuntu 20.04 and install it there?
Smth like
```
docker run -it --rm ubuntu:focal
Unable to find image 'ubuntu:focal' locally
focal: Pulling from library/ubuntu
1b9f3c55f9d4: Already exists 
Digest: sha256:8e5c4f0285ecbb4ead070431d29b576a530d3166df73ec44affc1cd27555141b
Status: Downloaded newer image for ubuntu:focal
root@38109811a41f:/# ls
bin  boot  dev  etc  home  lib  media  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
root@38109811a41f:/# 
```

carlosdelfino, 2025-01-02:
in ubuntu:focal

CMake Error at /opt/pico-sdk/tools/Findpicotool.cmake:7 (cmake_minimum_required):
  CMake 3.17 or higher is required.  You are running version 3.16.3
Call Stack (most recent call first):
  /opt/pico-sdk/tools/CMakeLists.txt:138 (find_package)
  /opt/pico-sdk/tools/CMakeLists.txt:485 (pico_init_picotool)
  /opt/pico-sdk/src/cmake/on_device.cmake:57 (picotool_postprocess_binary)
  CMakeLists.txt:147 (pico_add_extra_outputs)

carlosdelfino, 2025-01-03:
I had to download the cmake separately for it to work. I made a fork and updated the Readme.md: https://github.com/ArvoreDosSaberes/firmware-pi-rp2040/blob/main/README.md

Thanks for your help, I will have many questions from now on, as the Raspberry Pi and the Cortex-M architecture are a new world.

I hope I can count on the help of the team that developed this firmware.

closed 2025-01-03 by carlosdelfino

AIWintermuteAI, 2025-01-03:
> Raspberry Pi and the Cortex-M architecture are a new world.

Welcome to the brave new world then!

> I hope I can count on the help of the team that developed this firmware.

We can help with some issue, but adding WiFi will be quite a challenge. If you do manage to make it work, make sure to make a PR, we'll happily accept it.